nisrulz/sensey #5: Enable compatibility with emulator for running the sample app

nisrulz, 2016-05-10:
As of now the sample app would not return any data when running on an emulator. This is because possibly the emulator does not support any sensor mocking.

To do so a possible solution of interest I found over at this [link](http://stackoverflow.com/a/10677169/2745762).

ravidsrk, 2016-10-15:
@nisrulz Latest android emulator supports sensor mocking. https://twitter.com/molsjeroen/status/778163738285314048

nisrulz, 2016-10-15:
@ravidsrk Thanks for pointing that out. I will test and close this issue.

nisrulz, 2017-02-06:
I have tested this and the current version of Android Emulator with virtual sensors is able to run the library and the sample app without any modifications. 
Closing as works as intended.

closed 2017-02-06 by nisrulz

ravidsrk, 2017-02-06:
👍