nisrulz/sensey #6: Shake detection to sensitive

branemk, 2016-06-08:
Can we change sensitivity to shake detection ? Right now is to sensitive it detect when I pick up phone from table.

nisrulz, 2016-06-10:
@branemk Can you please explain the issue in a more verbose manner.

branemk, 2016-06-10:
@nisrulz I am talking about being able to set how much motion to detect. 

As you can see here https://github.com/nisrulz/sensey/blob/develop/library/src/main/java/com/github/nisrulz/sensey/ShakeDetector.java 

you have commented the line saying " Make this higher or lower according to how much
      motion you want to detect" , so I am asking for option to change that number.

nisrulz, 2016-06-10:
@branemk got it. I will provide an option to configure it in the next release.

nisrulz, 2016-06-14:
Fixed in 7c5e362b92133890e1798ef2beb451d23674a08a

closed 2016-06-14 by nisrulz