nisrulz/sensey #7: "Swipe left"/"Swipe right" does not work correctly

programmerr47, 2016-06-23:
Tested sample app on Nexus 5, Android 6.0. Try to swipe to right, it shows "Swipe left" label. Moreover when try to swipe to top or bottom, it shows "swipe left".

nisrulz, 2016-06-23:
The code is in `develop` branch, not fully functional. The final `release` will have it working.

programmerr47, 2016-06-23:
Hi. Thank you for your reply. I've already have, as I think, fix for that. And also I wrote tests for all classes. If you don't mind, I would like to make pull request.

nisrulz, 2016-06-24:
All pull request/contribution are welcome. Go ahead and send it.

nisrulz, 2016-06-24:
Code merged from pull request https://github.com/nisrulz/sensey/pull/8 , some fixes added and code cleanup done and new version is released at commit 32edad619d04ae79153d512a8506f4f58f3940f7

Closing this issue as the latest version `1.3.0` includes all fixes and solves this issue

closed 2016-06-24 by nisrulz