nisrulz/sensey #11: [Question] Enhancement of Javadocs

programmerr47, 2016-06-29:
Hi. Recently I saw your update and you added javadocs, and I think for library project it's awesome.
But may be it should be written more info than simple, for example,  "Creates new Flip Detector", because when you write `new FlipDetection(something)` you already now, that you creating new Detector. I think it would be greate if javadocs contained more specific info about, why we need, for example, such detector and e.t.c. What do you think about this idea?

nisrulz, 2016-06-29:
@programmerr47 Yes, I added the basic javadocs using a plugin which populates the javadoc comments . I will update it when I do get the time. Meanwhile all PR are welcome :)

programmerr47, 2016-06-29:
@nisrulz Sounds great. By the way I've created a pull request with fixing bug of leaking "default" detectors.

The special test was introduced in pull request. In short we twice calling for example "startProximityDetection" and then two times "stopProximityDetection". Then we will still have one detector registered in sensorManager. I fixed that.