nisrulz/sensey #13: How do we add the Touch listeners to specific views?

gouravd, 2016-07-06:
How do we add the Touch listeners to specific views?

nisrulz, 2016-07-07:
@gouravd I haven't tested the lib to work with specific views but I would test that out before I can make a statement about it.

k0shk0sh, 2016-10-14:
Override this in your custom view and you should be good to go.
```
@Override public boolean dispatchTouchEvent(MotionEvent event) {
    Sensey.getInstance().setupDispatchTouchEvent(event);
    return super.dispatchTouchEvent(event);
}
```

nisrulz, 2016-11-04:
@gouravd Can you confirm that this is resolved.

nisrulz, 2017-02-06:
Closing as already answered.

closed 2017-02-06 by nisrulz

gouravd, 2017-02-06:
I had to eventually create a wrapper view on top of my views and overridden the dispatch events for these wrapper views.