nisrulz/sensey #14: Proximity detection of a specific distance

MaximilianKeppeler, 2016-10-22:
We can work with the onNear and onFar event with current state of the library. 
In my use case, it would make sense to predefine a specific distance, where another event can be triggers, for example the Method can be called onCustomDistance. So the user can in my case input a distance between around 3 and 15 cm (just around, doesn't have to be exact) and I can use that event to trigger something specific in my app. 

I hope it's understandable. Thank you very much in advance. This library is really amazing - fantastic work!

NielsMasdorp, 2016-10-24:
`startProximityDetection` method can take a `float` threshold parameter. Isn't that what you mean?

MaximilianKeppeler, 2016-10-24:
Doesn't seem to have any influence on my device. I variegated with the value, but the threshold was still around 2 cm / as without any float value.

NielsMasdorp, 2016-10-24:
Weird, according to the Android Developer documentation the sensor usually returns the distance in cm so calling `startProximityDetection(10f, this)` should make it trigger at 10cm. Maybe try with a different device?

MaximilianKeppeler, 2016-10-24:
I only have one device, a Nexus 6P. With a float value of 10, the onFar Method is not even triggered anymore. (The onNear method is still triggered, but at the same distance, as without a threshold input at all.)

Which device do you have?

nisrulz, 2017-05-07:
Closing this issue as there isn't any direct possible solution to getting proximity updates from the proximity sensor.i.e it's a hardware limitation on what kind of proximity sensor a device has. 

A possible solution can be using the camera to measure the distance but that's a far-fetched solution taking into account what the goals of the library are now.

If in future it makes sense, will open up a new issue for the same.

closed 2017-05-07 by nisrulz