nisrulz/sensey #15: Chop Detection

indianpoptart, 2016-11-03:
Is there a way to add a chop detection like the Motorola Flashlight chop detection, or if sensey already does it. Thanks

nisrulz, 2016-11-04:
@indianpoptart Can you explain what you mean by "Chop" Detection.

indianpoptart, 2016-11-04:
Here's a short video:
[Chop Detection](https://www.youtube.com/watch?v=edtCMqK8Rn8)

nisrulz, 2017-02-13:
@indianpoptart  the latest release(1.6.0) has the option to detect chop gesture.

closing as implemented.

closed 2017-02-13 by nisrulz

rameshmkll, 2018-05-21:
Chop is not working
  Sensey.getInstance().init(DashBoardActivity.this);
chopListener=new ChopDetector.ChopListener() {
            @Override public void onChop() {
                // Chop gesture detected, do something
                Toast.makeText(mMyApp, "chop", Toast.LENGTH_LONG).show();
                Log.i("chop","started");
            }
        };
        Sensey.getInstance().startChopDetection(chopListener);