nisrulz/sensey #16: Detect Shake Stopped

Zulqurnain, 2016-12-15:
How to detect shake is stopped ?

nisrulz, 2016-12-15:
Can you be a bit clear in what exactly you want done ?

Just so you know its a single event that a shake is detected, and not that a shake has started and stopped. Such events can occur as multiple shakes detected but it is not detected as a stream of when a shake was started and when it was stopped. That is so because the detection works of by calculating the diff in acceleration and deceleration that happens and compares it for a threshold value. 

https://github.com/nisrulz/sensey/blob/develop/sensey/src/main/java/com/github/nisrulz/sensey/ShakeDetector.java#L59-L72

Zulqurnain, 2016-12-15:
in my case i'm doing something in my app , as long as user keep shaking its phone , but when it stop's it must revert back all operation what were done while shaking , now i understand how that , i propose there should be **onShakStop** function too in the interface which can detect the continues shaking have been stopped ?? 

And i hope it is not that difficult:

```
@Override
  protected void onSensorEvent(SensorEvent sensorEvent) {
    // Shake detection
    float x = sensorEvent.values[0];
    float y = sensorEvent.values[1];
    float z = sensorEvent.values[2];
    float mAccelLast = mAccelCurrent;
    mAccelCurrent = (float) Math.sqrt(x * x + y * y + z * z);
    float delta = mAccelCurrent - mAccelLast;
    mAccel = mAccel * 0.9f + delta;
    // Make this higher or lower according to how much
    // motion you want to detect
    if (mAccel > threshold) {
      shakeListener.onShakeDetected();
    }else{
      // shaking is stopped do something about it ???
   }
  }
```

nisrulz, 2017-02-13:
@Zulqurnain the latest release(1.6.0) the option to detect when shake gesture is stopped.

Closing as fixed.

closed 2017-02-13 by nisrulz