nisrulz/sensey #17: onFar is triggered when switching to Android N Multi View.

MaximilianKeppeler, 2016-12-27:
That's some odd issue I have at the moment. The onFar method (from ProximityListener) is triggered when I switch to the Android N multi view. 

`Sensey.getInstance().init(this);
        proximityListener = new ProximityDetector.ProximityListener() {
            @Override
            public void onNear() {
                Log.d("Test", "onNear: ");
            }

            @Override
            public void onFar() {
                Log.d("Test", "onFar: ");
                triggerOnClick();
            }
        };
        Sensey.getInstance().startProximityDetection(proximityListener);`

MaximilianKeppeler, 2016-12-27:
It's replicable with your sample. Just tried it out.

nisrulz, 2018-08-18:
@MaximilianKeppeler I have restarted working on my library.  Is this still an issue, if yes can you share a video link for me to see how you are able to replicate it?