nisrulz/sensey #18: Is it possible to create more than one Sensey instance in an activity listening on different views

c0state, 2017-01-24:
It seems like only one instance of Sensey can be created per activity (or context to be precise). Is there any way to create and attach different Sensey listeners to more than one view in a single activity?

nisrulz, 2017-02-06:
@c0state the reason Sensey uses a singleton , is because it access sensor data which are specifically hardware that you have access to. The data stream you would usually get would also be one, as the hardware is going to be one. Hence even if you had multiple listeners you would still be accessing the same data stream. In this scenario all your listeners would still fire up even if you used multiple Sensey instances.

As of current state, Sensey cannot get data stream from multiple sensors of the same type in the device, which is when this would be useful, wherein Sensey could access 2 different sensors and respond on each as the data changes. 

Let me know what are your thoughts on this if this makes sense to you.

c0state, 2017-04-30:
That makes sense, thanks for the detailed information.

closed 2017-04-30 by c0state