nisrulz/sensey #20: Need a single event for Flip detection

aritraroy, 2017-02-17:
I just saw your library, actually liked it and thought to give it a try.

After attaching a flip detection listener in my activity, I see that the events are continuously getting fired. Like,

```
private FlipDetector.FlipListener mFlipListener = new FlipDetector.FlipListener() {
        @Override
        public void onFaceUp() {
            // Device Facing up
            Timber.d("DEVICE UP");
        }

        @Override
        public void onFaceDown() {
            // Device Facing down
            Timber.d("DEVICE DOWN");
        }
    };
```
in this code, I am getting continuous logs for "DEVICE UP" when the device is facing up and continuous logs for "FACE DOWN" when the device is facing down.

But my requirement is something different. If the device is facing up, it should just give one single onFaceUp() callback, and if the device is facing down it should just give one single onFaceDown() callback.

Is it possible with your library? If not, I will have to implement it myself. Just wanted to check if your library can do that or if you are willing to add it as an enhancement.

nisrulz, 2017-02-17:
it's not possible as per the current release, but I can push it out in the next release.

aritraroy, 2017-02-17:
Okay. Thanks. Any tentative date for the next release?

nisrulz, 2017-02-17:
I am planning to do it week by week basis i.e will try to push the next release this weekend.

If you think you can can fix it yourself and would like to submit a PR, go ahead with that.

nisrulz, 2017-02-28:
@aritraroy I have updated the code and the new release fixes this issue. Please confirm and close this issue if it works as expected.

aritraroy, 2017-03-01:
Thank you, I will give it a try as soon as possible and let you know.

closed 2017-03-19 by aritraroy