nisrulz/sensey #21: Attempt to invoke virtual method 'float android.view.MotionEvent.getX()' on a null object reference

PedroBarbosaPT, 2017-02-18:
It works, but sometimes it shows this error:
I don't know how to replicate tho :/

```
Process: com.test.testmenu, PID: 5675
                  java.lang.NullPointerException: Attempt to invoke virtual method 'float android.view.MotionEvent.getX()' on a null object reference
                      at com.github.nisrulz.sensey.TouchTypeDetector$a.onScroll(SourceFile:169)
                      at android.view.GestureDetector.onTouchEvent(GestureDetector.java:619)
                      at android.support.v4.view.GestureDetectorCompat$GestureDetectorCompatImplJellybeanMr2.onTouchEvent(GestureDetectorCompat.java:479)
                      at android.support.v4.view.GestureDetectorCompat.onTouchEvent(GestureDetectorCompat.java:542)
                      at com.github.nisrulz.sensey.Sensey.setupDispatchTouchEvent(SourceFile:4102)
                      at com.quarkprojects.detetordeazeite.MainActivity.dispatchTouchEvent(MainActivity.java:1202)
                      at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:71)
                      at android.support.v7.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:71)
                      at com.android.internal.policy.PhoneWindow$DecorView.dispatchTouchEvent(PhoneWindow.java:2364)
                      at android.view.View.dispatchPointerEvent(View.java:9514)
                      at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:4230)
                      at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:4096)
                      at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3642)
                      at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3695)
                      at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3661)
                      at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:3787)
                      at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3669)
                      at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:3844)
                      at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3642)
                      at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:3695)
                      at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:3661)
                      at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:3669)
                      at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:3642)
                      at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:5922)
                      at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:5896)
                      at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:5857)
                      at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:6025)
                      at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:185)
                      at android.os.MessageQueue.nativePollOnce(Native Method)
                      at android.os.MessageQueue.next(MessageQueue.java:323)
                      at android.os.Looper.loop(Looper.java:135)
                      at android.app.ActivityThread.main(ActivityThread.java:5417)
                      at java.lang.reflect.Method.invoke(Native Method)
                      at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
                      at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)
```

nisrulz, 2017-02-18:
@PedroBarbosaPT  Can you provide device details and os details.

PedroBarbosaPT, 2017-02-18:
@nisrulz Android Emulator - Nexus 5 API 23

nisrulz, 2017-02-28:
@PedroBarbosaPT I have yet to reproduce this issue. Just so you know this issue isn't abandoned. 😅  Will update you when I am able to replicate and provide a possible solution.

PedroBarbosaPT, 2017-02-28:
@nisrulz No problem. The error never happened again, so ... 😅

nisrulz, 2017-02-28:
Hmm.. good to know that. I am closing this issue until it resurfaces again.

closed 2017-02-28 by nisrulz

Patra89, 2017-11-10:
I have the same problem. From my stats on Fabric it looks like it occurs only on Samsung (94%) and Huawei(6%) devices. I wasn't able to replicate the problem myself.  It's the most occurring bug in my app right now, Is there any solution to this?

nisrulz, 2017-11-10:
Hi @Patra89 
Could you share information from your fabric account. For both devices where you are seeing this. Also please provide me full details of the device specs. If you do figure out how to replicate this that would be really helpful too.

Patra89, 2017-11-10:
**Stats from Fabric:**

**Samsung devices:** Galaxy S4, SM-G950U, Galaxy S7 Edge, SM-G950F, SM-A320FL, Galaxy S7, Galaxy S6, SM-N950F, SM-N950F, Galaxy Note4, Galaxy S6 Edge, Galaxy A7(2016), Galaxy S6 Edge+

**Huawei devices:** P9, P8 Lite

**Operating systems:**
7 - 64%
5 - 23%
6 - 13%
(I don't support anything below 5.0)

**Device statistics:**
Proximity On - 6%
App in Focus - 6%
Rooted - 0%
____________________________________________________

**Raport on the same problem from my Google Play Console:**

**Devices:**
Galaxy S4 (jflte) | 23,5%
Galaxy S8 (dreamqltesq) |  11,8%
Galaxy S7 Edge (hero2lte) | 8,8%
Galaxy A3 (2017) (a3y17lte) | 8,8%
Galaxy S7 (heroqltevzw) | 5,9%
Galaxy S7  (heroqltetmo) | 5,9%
Galaxy S7 Edge (hero2qltetmo) | 5,9%
Galaxy S8 (dreamlte) | 5,9%
Galaxy S6 (zeroflte) |  2,9%
Galaxy S6 Edge+ (zenltetmo) | 2,9%
Galaxy Note4 (trlte) | 2,9%
Galaxy S7  (herolte) | 2,9%
Galaxy Note8 (greatlte) | 2,9%
Galaxy S8+ (dream2qltecan) | 2,9%
Galaxy A7(2016) (a7xeltextc) | 2,9%
Galaxy A7(2016) (a7xelte)

**Operating systems:**

Android 7.0 | 58,8%
Android 5.0 | 23,5%
Android 6.0 | 14,7%
Android 7.1 | 2,9%

VanFanelia, 2020-10-14:
Have the same bug here on my test Device.

Stack trace:

2020-10-14 15:07:32.521 18965-18965/de.sipgate.app.satellite V/ProfileListEntry: [2020-10-14 15:07:32:517] [18965] [VERBO] [ProfileListEntry]: Event was ACTION_MOVE
2020-10-14 15:07:32.528 18965-18965/de.sipgate.app.satellite V/ProfileListEntry: [2020-10-14 15:07:32:528] [18965] [VERBO] [ProfileListEntry]: Data: rawX: 486.82617, x: 486.82617, xPrecision: 1.0 
2020-10-14 15:07:32.568 18965-18965/de.sipgate.app.satellite V/ProfileListEntry: [2020-10-14 15:07:32:567] [18965] [VERBO] [ProfileListEntry]: Event was ACTION_CANCEL
2020-10-14 15:07:32.575 18965-18965/de.sipgate.app.satellite V/ProfileListEntry: [2020-10-14 15:07:32:574] [18965] [VERBO] [ProfileListEntry]: Your time Delay was: 131 and your distance: -395.8642
2020-10-14 15:07:32.587 18965-18965/de.sipgate.app.satellite V/ProfileListEntry: [2020-10-14 15:07:32:586] [18965] [VERBO] [ProfileListEntry]: Reset and Start animation
2020-10-14 15:07:32.596 18965-18965/de.sipgate.app.satellite I/ProfileListEntry: [2020-10-14 15:07:32:595] [18965] [INFO]  [ProfileListEntry]: I choose this was a click
2020-10-14 15:07:32.631 18965-18965/de.sipgate.app.satellite D/ProfileRepository: [2020-10-14 15:07:32:630] [18965] [DEBUG] [ProfileRepository]: attempt to parse 11246ada-4f80-4afb-b613-66d75f00ca29
2020-10-14 15:07:32.633 18965-18965/de.sipgate.app.satellite W/ClassMapper: No setter/field for trigger found on class de.sipgate.app.satellite.repository.Profile
2020-10-14 15:07:32.635 18965-18965/de.sipgate.app.satellite D/ProfileRepository: [2020-10-14 15:07:32:634] [18965] [DEBUG] [ProfileRepository]: attempt to parse b31906d6-5ee9-4839-ae4e-885fbc62c95f
2020-10-14 15:07:32.636 18965-18965/de.sipgate.app.satellite W/ClassMapper: No setter/field for trigger found on class de.sipgate.app.satellite.repository.Profile
2020-10-14 15:07:32.674 18965-18965/de.sipgate.app.satellite E/InputEventReceiver: Exception dispatching input event.
2020-10-14 15:07:32.674 18965-18965/de.sipgate.app.satellite D/AndroidRuntime: Shutting down VM
2020-10-14 15:07:32.686 18965-18965/de.sipgate.app.satellite E/AndroidRuntime: FATAL EXCEPTION: main
    Process: de.sipgate.app.satellite, PID: 18965
    java.lang.NullPointerException: Attempt to invoke virtual method 'float android.view.MotionEvent.getX()' on a null object reference
        at com.github.nisrulz.sensey.TouchTypeDetector$a.onScroll(SourceFile:84)
        at android.view.GestureDetector.onTouchEvent(GestureDetector.java:651)
        at androidx.core.view.GestureDetectorCompat$GestureDetectorCompatImplJellybeanMr2.onTouchEvent(GestureDetectorCompat.java:480)
        at androidx.core.view.GestureDetectorCompat.onTouchEvent(GestureDetectorCompat.java:543)
        at com.github.nisrulz.sensey.Sensey.setupDispatchTouchEvent(SourceFile:1240)
        at de.sipgate.app.satellite.main.MainActivity.dispatchTouchEvent(MainActivity.kt:326)
        at androidx.appcompat.view.WindowCallbackWrapper.dispatchTouchEvent(WindowCallbackWrapper.java:69)
        at com.android.internal.policy.DecorView.dispatchTouchEvent(DecorView.java:514)
        at android.view.View.dispatchPointerEvent(View.java:12877)
        at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent(ViewRootImpl.java:5771)
        at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess(ViewRootImpl.java:5535)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4983)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5036)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:5002)
        at android.view.ViewRootImpl$AsyncInputStage.forward(ViewRootImpl.java:5157)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:5010)
        at android.view.ViewRootImpl$AsyncInputStage.apply(ViewRootImpl.java:5214)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4983)
        at android.view.ViewRootImpl$InputStage.onDeliverToNext(ViewRootImpl.java:5036)
        at android.view.ViewRootImpl$InputStage.forward(ViewRootImpl.java:5002)
        at android.view.ViewRootImpl$InputStage.apply(ViewRootImpl.java:5010)
        at android.view.ViewRootImpl$InputStage.deliver(ViewRootImpl.java:4983)
        at android.view.ViewRootImpl.deliverInputEvent(ViewRootImpl.java:7803)
        at android.view.ViewRootImpl.doProcessInputEvents(ViewRootImpl.java:7768)
        at android.view.ViewRootImpl.enqueueInputEvent(ViewRootImpl.java:7726)
        at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent(ViewRootImpl.java:8007)
        at android.view.InputEventReceiver.dispatchInputEvent(InputEventReceiver.java:202)
        at android.view.InputEventReceiver.nativeConsumeBatchedInputEvents(Native Method)
        at android.view.InputEventReceiver.consumeBatchedInputEvents(InputEventReceiver.java:191)
        at android.view.ViewRootImpl.doConsumeBatchedInput(ViewRootImpl.java:7917)
        at android.view.ViewRootImpl$ConsumeBatchedInputRunnable.run(ViewRootImpl.java:8061)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1092)
        at android.view.Choreographer.doCallbacks(Choreographer.java:893)
        at android.view.Choreographer.doFrame(Choreographer.java:806)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1078)
        at android.os.Handler.handleCallback(Handler.java:907)
[... cut ...]  there a still running some event calculation in background. 

Heres a video when i try to recreate the bug: 
https://drive.google.com/file/d/1-CQRggoRcw0b1GqTD1lHIIOGpMLaOhUn/view?usp=sharing

I will try to fix it with try Catch in my MainActivity:

`    override fun dispatchTouchEvent(event: MotionEvent?): Boolean {
        // Setup onTouchEvent for detecting type of touch gesture
        Sensey.getInstance().setupDispatchTouchEvent(event)
        return super.dispatchTouchEvent(event)
    }
`

VanFanelia, 2020-10-14:
My temporary fix is a big try catch around sensey's setup method. It works, but is not a reliable solution.

`
override fun dispatchTouchEvent(event: MotionEvent?): Boolean {
        // Setup onTouchEvent for detecting type of touch gesture
        try {
            Sensey.getInstance().setupDispatchTouchEvent(event)
        } catch (e: Exception){
            Timber.e(e, "Sensey got an exception. try to ignore it")
        }
        return super.dispatchTouchEvent(event)
    }
`

OEssHRERLr770rGGLOH0, 2021-08-10:
I don't know if it's related but I've gotten a couple of _somewhat_ similar crashes. Please let me know if you'd prefer that UI open a new issue for it. See below:

```
java.lang.NullPointerException: 
  at com.github.nisrulz.sensey.TouchTypeDetector$a.onScroll (SourceFile:1)
  at android.view.GestureDetector.onTouchEvent (GestureDetector.java:650)
  at androidx.core.view.GestureDetectorCompat$GestureDetectorCompatImplJellybeanMr2.onTouchEvent (GestureDetectorCompat.java:480)
  at androidx.core.view.GestureDetectorCompat.onTouchEvent (GestureDetectorCompat.java:543)
  at com.github.nisrulz.sensey.TouchTypeDetector.a (SourceFile:10)
  at com.github.nisrulz.sensey.Sensey.setupDispatchTouchEvent (SourceFile:2)
  at com.tobianoapps.timerise.ui.MainActivity.dispatchTouchEvent (MainActivity.kt:146)
  at androidx.appcompat.view.WindowCallbackWrapper.dispatchTouchEvent (WindowCallbackWrapper.java:69)
  at com.android.internal.policy.DecorView.dispatchTouchEvent (DecorView.java:692)
  at android.view.View.dispatchPointerEvent (View.java:13721)
  at android.view.ViewRootImpl$ViewPostImeInputStage.processPointerEvent (ViewRootImpl.java:6197)
  at android.view.ViewRootImpl$ViewPostImeInputStage.onProcess (ViewRootImpl.java:5975)
  at android.view.ViewRootImpl$InputStage.deliver (ViewRootImpl.java:5424)
  at android.view.ViewRootImpl$InputStage.onDeliverToNext (ViewRootImpl.java:5477)
  at android.view.ViewRootImpl$InputStage.forward (ViewRootImpl.java:5443)
  at android.view.ViewRootImpl$AsyncInputStage.forward (ViewRootImpl.java:5602)
  at android.view.ViewRootImpl$InputStage.apply (ViewRootImpl.java:5451)
  at android.view.ViewRootImpl$AsyncInputStage.apply (ViewRootImpl.java:5659)
  at android.view.ViewRootImpl$InputStage.deliver (ViewRootImpl.java:5424)
  at android.view.ViewRootImpl$InputStage.onDeliverToNext (ViewRootImpl.java:5477)
  at android.view.ViewRootImpl$InputStage.forward (ViewRootImpl.java:5443)
  at android.view.ViewRootImpl$InputStage.apply (ViewRootImpl.java:5451)
  at android.view.ViewRootImpl$InputStage.deliver (ViewRootImpl.java:5424)
  at android.view.ViewRootImpl.deliverInputEvent (ViewRootImpl.java:8501)
  at android.view.ViewRootImpl.doProcessInputEvents (ViewRootImpl.java:8421)
  at android.view.ViewRootImpl.enqueueInputEvent (ViewRootImpl.java:8374)
  at android.view.ViewRootImpl$WindowInputEventReceiver.onInputEvent (ViewRootImpl.java:8616)
  at android.view.InputEventReceiver.dispatchInputEvent (InputEventReceiver.java:198)
  at android.view.InputEventReceiver.nativeConsumeBatchedInputEvents (Native Method)
  at android.view.InputEventReceiver.consumeBatchedInputEvents (InputEventReceiver.java:187)
  at android.view.ViewRootImpl.doConsumeBatchedInput (ViewRootImpl.java:8575)
  at android.view.ViewRootImpl$ConsumeBatchedInputRunnable.run (ViewRootImpl.java:8643)
  at android.view.Choreographer$CallbackRecord.run (Choreographer.java:988)
  at android.view.Choreographer.doCallbacks (Choreographer.java:765)
  at android.view.Choreographer.doFrame (Choreographer.java:694)
  at android.view.Choreographer$FrameDisplayEventReceiver.run (Choreographer.java:967)
  at android.os.Handler.handleCallback (Handler.java:873)
  at android.os.Handler.dispatchMessage (Handler.java:99)
  at android.os.Looper.loop (Looper.java:214)
  at android.app.ActivityThread.main (ActivityThread.java:7156)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:494)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:975)
```

Device was a Samsung Galaxy J7 Neo running Android 9 (SDK 28).

Any idea how I could investigate the source of this error? Thanks!