nisrulz/sensey #27: The shake mode is too sensitive to me

drogbababa, 2017-04-07:
Please fill out the blanks below describing your issue

What are the steps to reproduce this issue?
-------------------------------------------
1. …
2. …
3. …
…

What happens?
some little movement is on shake.

What were you expecting to happen?
i expect the shakeThreshold can be larger.or users can modify it themselves

Any logs, error output, bugreport etc?
----------------------------
(If it’s long, please paste to https://ghostbin.com/ and insert the link here.)

Any other comments?
-------------------
…

What versions of software are you using?
----------------------------------------
**Device Information:** …

**Android Version:** …

**Configuration Information:** …

[More related information if you have can provide]

nisrulz, 2017-04-07:
You can set the threshold value when starting the shake detector.

Check here under the wiki.

https://github.com/nisrulz/sensey/wiki/Usage#shake

gauravat16, 2017-05-06:
@nisrulz Can you tell me the units and scale for Threshold and time?

nisrulz, 2017-05-07:
Threshold here is acceleration i.e m/s*s
timeBeforeDeclaringShakeStopped is the time in milliseconds.

I would update the same in the wiki docs, a new version of Sensey is under development and I would make sure more documentation and other details are available by the time it releases.