nisrulz/sensey #28: Feature Request: Tap on device body detection.

bjaanes, 2017-05-18:
Some devices don't actually have screens or are a hand-held device. I am thinking of smart glasses with android. I have an ODG R-7 and I want an easy way to start voice detection, and one of the easiest way I have thought if to double-tap on the body of the glasses.

Would it be much work to add some sort of body-tap detection to this library?

nisrulz, 2017-05-18:
@bjaanes That is an interesting thought. I am currently working on a major update to my library (v2.0) and I can definitely look into this feature.

I would, however, need to look at the feasibility of this in the first place. Also since I do not have any such hardware would you be interested in testing it out when I have the code for it on your device?

bjaanes, 2017-05-22:
Sure, that makes sense. I would be very much interested in trying that out on my pair of ODG R-7! :)