nisrulz/sensey #29: How to implement dispatchTouchEvent() in fragment?

mistrydarshan99, 2017-06-12:
Hello, 

This library works very well in activity but not getting touch detection in the fragment.

So how to implement in the fragment. And also don't get dispatchTouchEvent() method in a fragment.

yesidlazaro, 2017-07-30:
any news here?

pnkram, 2018-07-06:
So it's not possible at the moment to implement touch detection inside a fragment? I would need put 2 square/rectangle inside an activity and put 2 different touch detector.

pnkram, 2018-07-06:
Hi, inside my mainactivity I put a FrameLayout in which there is another layout. Inside that layout I initialized sensey while in mainactivity i set dispatchTouchEvent like this:

```
    @Override
    public boolean dispatchTouchEvent(MotionEvent event) {
        // Setup onTouchEvent for detecting type of touch gesture
        //Sensey.getInstance().setupDispatchTouchEvent(event);
        rightFragment.sensey.getInstance().setupDispatchTouchEvent(event);
        return super.dispatchTouchEvent(event);
    }
```
However when I touch the screen, the touch event is being recognized in everywhere while it should recognized only inside the layout inside the fragment. 

Can anyone please help me?