nisrulz/sensey #30: Multiple Detection

indianpoptart, 2017-07-13:
Is there a way to detect multiple consecutive motions? Such as detecting two chops in ChopListener, as opposed to a singular chop.

Thanks

indianpoptart, 2017-07-13:
Additional info:
Would I have to just make another listener inside onChop? or is that a dumb idea. Thanks

nisrulz, 2017-07-13:
Well you can count the number of times a chop event is triggered. Its not a dumb idea, but since its not built into the lib as of now, doing it that way should work. It isn't the best solution though.

I am just wondering how much of a use case there is to put a counter in the lib itself. The thing is, all these events are time based. So the counter eventually would need to reset if the chop event didnot happen within a certain time duration. 
 
Providing it internal to lib is possible and I can add it to the upcoming features to develop, if its important enough.

Let me know your thoughts.

indianpoptart, 2017-07-14:
I can probably just implement a counter, didn't even think of that :)

You don't have to add it to the lib, up to you though. Probably easier to add a little not to the readme or wiki.

Thanks a bunch!

nisrulz, 2017-07-14:
I guess this solves your issue at hand. I will open a new issue if I see it fit to include it in the future.

Closing the issue for now.

closed 2017-07-14 by nisrulz