nisrulz/sensey #31: Not Working in Android 7.0 but Touch Detectors are working

PankajRaincan, 2017-07-28:
**Do you want to request a *feature* or report a *bug*?**

**What is the current behavior?**

**If the current behavior is a bug, please provide the steps to reproduce.**

**Any logs, error output, bugreport etc?**

**What is the expected behavior?**

**Any other comments?**

**What versions of software are you using?**
+ **Device Information:** 

+ **Android Version:** 

+ **Configuration Information:** 

+ **Misc:**

nisrulz, 2017-07-28:
@PankajRaincan can you explain "Not Working in Android 7.0" a bit more clear as to what is not working?

PankajRaincan, 2017-07-29:
Hey,
Whenever I'm trying to do any motion gestures its not responding at all.Am I doing anything wrong as I tried it with Activity as well as Service.And even I tried with your sample application still no luck.

`Sensey.getInstance().init(this);
Sensey.getInstance().startWaveDetection(this);
 @Override
  public void onWave() {
    //my logs
  }
`

nisrulz, 2017-07-29:
Wave detection works using Proximity sensor, I am assuming you are not in the proximity of your device. Are you having issues with other event types too ? If so which ones. Its hard for me to debug this, if you cannot provide me more info apart from saying its not working. If possible hook your device to a laptop and gather whatever your logcat is able to log from the system. 

Another thing to check up would be about the functionality of the sensors, which you can test by downloading any sensor app and verify they are functional.

Let me if you have information regarding that.

PankajRaincan, 2017-07-29:
Proximity Sesnor is functional I have checked it by putting logs in onSensorChanged callback.

> 07-29 15:06:02.294 7573-7573/? I/art: Late-enabling -Xcheck:jni
07-29 15:06:02.561 7573-7573/com.github.nisrulz.senseysample W/System: ClassLoader referenced unknown path: /data/app/com.github.nisrulz.senseysample-2/lib/arm
07-29 15:06:02.586 7573-7573/com.github.nisrulz.senseysample I/InstantRun: starting instant run server: is main process
07-29 15:06:02.587 7573-7573/com.github.nisrulz.senseysample V/InstantRun: Starting server socket listening for package com.github.nisrulz.senseysample on android.net.LocalSocketAddress@777df7a
07-29 15:06:02.588 7573-7573/com.github.nisrulz.senseysample V/InstantRun: Started server for package com.github.nisrulz.senseysample
07-29 15:06:02.756 7573-7573/com.github.nisrulz.senseysample W/art: Before Android 4.1, method android.graphics.PorterDuffColorFilter android.support.graphics.drawable.VectorDrawableCompat.updateTintFilter(android.graphics.PorterDuffColorFilter, android.content.res.ColorStateList, android.graphics.PorterDuff$Mode) would have incorrectly overridden the package-private method in android.graphics.drawable.Drawable
07-29 15:06:03.037 7573-7730/com.github.nisrulz.senseysample I/Adreno: QUALCOMM build                   : 7d18700, I8ee426a9a2
                                                                       Build Date                       : 10/07/16
                                                                       OpenGL ES Shader Compiler Version: XE031.09.00.03
                                                                       Local Branch                     : mybranch22308589
                                                                       Remote Branch                    : quic/LA.BR.1.3.6_rb1.6
                                                                       Remote Branch                    : NONE
                                                                       Reconstruct Branch               : NOTHING
07-29 15:06:03.043 7573-7730/com.github.nisrulz.senseysample I/OpenGLRenderer: Initialized EGL, version 1.4
07-29 15:06:03.043 7573-7730/com.github.nisrulz.senseysample D/OpenGLRenderer: Swap behavior 1

these are the logs apart from this nothing is being listed in my logcat.

nisrulz, 2017-07-29:
Ok. So from this what I can guess is that everything works for you except the Wave detection, correct me if I am wrong.

Now if the above case is the issue, then  maybe you are doing the gesture wrong. So the way it is supposed to work is you move your hand over your proximity center and the time diff is calculated , which if it exceeds the threshold, it is detected as a wave gesture event. 

I am actually working on pushing out some videos showing how such gestures are supposed to be done and what sensey supports.

PankajRaincan, 2017-07-29:
Yes that might be the case though I have tried many variations of wave. I hope as soon as your videos come my problem will get solved.
Thanks