nisrulz/sensey #32: New permission added: RECORD_AUDIO. Is it possible to remove it?

amanzan, 2017-10-03:
I added this library to my project, and when trying to upload the apk to Google Play, I get the following warning: 
WARNING
Users that have the APK with version code 10 may need to accept the android.permission.RECORD_AUDIO permission, which may result in them not upgrading to this version of the app.

I assume this is for the "Sound Level Detection" and I see this permission is added to the library's AndroidManifest. Would it be possible to remove it? I think it's a better option that whoever needs this feature, adds this permission by himself/herself to avoid warnings even when not needed

nisrulz, 2017-10-03:
Its actually been removed in the `develop` branch. I just haven't released it in the `master` branch as a release.

amanzan, 2017-10-04:
Ah ok, thanks! When are you planning on releasing the new version?

nisrulz, 2017-10-04:
I think I can release it by this weekend.

martinflorek, 2017-11-26:
@nisrulz anything new about this issue? and thanks for this great library!

nisrulz, 2017-11-27:
@martinflorek I had been traveling. I have been working on one version of this library which is a major update, but I see the permission problem. I will release a version without the permission asap and then move to the next major release.

Please give me sometime, possibly a day or two and I will surely push out a version so that the permission issue is solved.

Thank you for your patience and using this library.

martinflorek, 2017-11-27:
@nisrulz thanks for the quick reply and for the hard work.

I needed to release a beta version of my app asap, so I have just removed the permission from my app via Gradle. Hope it will not cause issues in release build (I am using just the shake detector) and I will wait for your stable release for my stable release. Thank you again.

nisrulz, 2017-12-05:
@martinflorek I have just released a new version of Sensey (v.18.0), which addresses this issue. Please checkout the changelog for breaking changes in this release [here](https://github.com/nisrulz/sensey/releases/tag/1.8.0)

Apologies for the delayed update for this bug.
Thank you for your patience and using this library.

> Please close this issue if your issue is resolved.

martinflorek, 2017-12-05:
@nisrulz Thank you! I will test it tonight. My app is targeting API 19 and up so there should not be a problem.

martinflorek, 2017-12-06:
@nisrulz Works good, issue can be closed. Thank you.

closed 2017-12-07 by nisrulz