nisrulz/sensey #36: Step Detection

javadVatan, 2018-01-03:
Hi dear.
 I read in following link  about step Detection, but it's not really in your library!!
Did you removed this feature?
https://github.com/nisrulz/sensey/wiki/Usage#steps   --> here you talked about it.

nisrulz, 2018-01-03:
No, I didnot remove it. Its a mistake, I did not notice it. Thank you for reporting it. 
Actually I added a few more detectors but then found a critical bug in them and pulled down the release. I will fix the readme for now and the wiki page, until it is fixed and properly released.

I am actually working a bigger update to Sensey and plan to release more detectors along with it.

Thank you for reporting it, I am on it 😅

javadVatan, 2018-01-04:
okay, thank you. 
It was so fun for me. I checked all version of your library for find them 😅😅

closed 2018-01-04 by javadVatan