nisrulz/sensey #39: Gesture RecognitionApp

tdubes, 2018-03-18:
**Do you want to request a *feature* or report a *bug*?**
      Report a bug
**What is the current behavior?**
     Using the library l would like to use touch based gesture,  so am having errors within the method that        is said to be included as said by the documentation and it is said to be very important.
it is saying: Cannot resolve method 'dispatchTouchEvent'(android.view.MotionEvent)

**If the current behavior is a bug, please provide the steps to reproduce.**
  This is a link to a public repository:   https://github.com/tdubes/Gesture-RecognitionApp

**Any logs, error output, bugreport etc?**

**What is the expected behavior?**
The method is supposed to start  intercept touch actions in activity by overriding the  dispatchTouchEvent.
**Any other comments?**

**What versions of software are you using?**
+ **Device Information:** 
  

Nexus 5X API 26 4 |   | 1080 × 1920: 420dpi | 26 | Android 8.0 (Google APIs) | x86 | 1 GB |
+ **Android Version:** 
              Android 3.0.1

tdubes, 2018-03-18:
Please Help because am using this library for project and l am still new to android.

nisrulz, 2018-03-19:
@tdubes I see no code in the public repo. Please define the steps to reproduce. I did check it in the activity and it works, you should check the sample app for implementation once.