nisrulz/sensey #41: Request for 2.0

rgocal, 2018-06-17:
I'm using Sensey as the gesture driver in my launcher. It works beautifully but I'd like to see more finger gestures than hardware. One finger clicks and movements, 2 finger clicks and movement, 3 finger clicks and movement.

Idk if this would be considered a gesture but could a retinal sensor detection be implemented? Like for winks, blinking? Probably not supported on all devices.

Fingure print readers would be interesting to see.

Of all other things, this is perfect. Keep it up!

nisrulz, 2018-06-17:
@rgocal Thank you for your words, feedback and feature suggestions. I am trying to make Sensey more modular and the new version 2.0.0 is planned but without a timeline. 

I will keep your feature request in mind and release it if it makes sense with the new release. 

On a side note, if you would like me to mention your launcher in the Readme of this repo, do share with me the link and I will do so. 

Thank you once again for using Sensey and having faith in the project. Your words mean a lot when it comes to motivation to keep on working on this project 😸

rgocal, 2018-06-17:
Once my launcher update hits an open beta and I feel comfortable with its performance, il drop a link!

rgocal, 2018-12-17:
Feeling abit xomfortable to share this now. Working on making sime Sensey enhancments in my launcher soon. Your more tham welcomed to use this in your readme as an example ofbusage if you approve. 

https://play.google.com/store/apps/details?id=com.gsd.launcher.sapphyx

rgocal, 2019-05-02:
Any updates on the development of 2.0?