nisrulz/sensey #45: App keeps crashing

debmalya191, 2018-10-23:
My github rep-
https://github.com/debmalya191/gestureapp

I am making a simple app where on the shake gesture a new activity opens from the main activity but the app keeps crashing.

Here is my logcat-

dRuntime: FATAL EXCEPTION: main
                                                                             Process: com.example.mahe.gestureapp, PID: 28290
                                                                             java.lang.NoSuchMethodError: No static method getFont(Landroid/content/Context;ILandroid/util/TypedValue;ILandroid/widget/TextView;)Landroid/graphics/Typeface; in class Landroid/support/v4/content/res/ResourcesCompat; or its super classes (declaration of 'android.support.v4.content.res.ResourcesCompat' appears in /data/app/com.example.mahe.gestureapp-rPGMtdXCLHaxk64RVyNspA==/split_lib_dependencies_apk.apk)
                                                                                 at android.support.v7.widget.TintTypedArray.getFont(TintTypedArray.java:119)
                                                                                 at android.support.v7.widget.AppCompatTextHelper.updateTypefaceAndStyle(AppCompatTextHelper.java:208)
                                                                                 at android.support.v7.widget.AppCompatTextHelper.loadFromAttributes(AppCompatTextHelper.java:110)
                                                                                 at android.support.v7.widget.AppCompatTextHelperV17.loadFromAttributes(AppCompatTextHelperV17.java:38)
                                                                                 at android.support.v7.widget.AppCompatTextView.<init>(AppCompatTextView.java:81)
                                                                                 at android.support.v7.widget.AppCompatTextView.<init>(AppCompatTextView.java:71)
                                                                                 at android.support.v7.widget.AppCompatTextView.<init>(AppCompatTextView.java:67)
                                                                                 at android.support.v7.widget.Toolbar.setTitle(Toolbar.java:753)
                                                                                 at android.support.v7.widget.ToolbarWidgetWrapper.setTitleInt(ToolbarWidgetWrapper.java:261)
                                                                                 at android.support.v7.widget.ToolbarWidgetWrapper.setWindowTitle(ToolbarWidgetWrapper.java:243)
                                                                                 at android.support.v7.widget.ActionBarOverlayLayout.setWindowTitle(ActionBarOverlayLayout.java:621)
                                                                                 at android.support.v7.app.AppCompatDelegateImplV9.onTitleChanged(AppCompatDelegateImplV9.java:631)
                                                                                 at android.support.v7.app.AppCompatDelegateImplV9.ensureSubDecor(AppCompatDelegateImplV9.java:328)
                                                                                 at android.support.v7.app.AppCompatDelegateImplV9.setContentView(AppCompatDelegateImplV9.java:284)
                                                                                 at android.support.v7.app.AppCompatActivity.setContentView(AppCompatActivity.java:139)
                                                                                 at com.example.mahe.gestureapp.MainActivity.onCreate(MainActivity.java:15)
                                                                                 at android.app.Activity.performCreate(Activity.java:6984)
                                                                                 at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1235)
                                                                                 at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2783)
                                                                                 at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2909)
                                                                                 at android.app.ActivityThread.-wrap11(Unknown Source:0)
                                                                                 at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1606)
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:105)
                                                                                 at android.os.Looper.loop(Looper.java:164)
                                                                                 at android.app.ActivityThread.main(ActivityThread.java:6592)
                                                                                 at java.lang.reflect.Method.invoke(Native Method)
                                                                                 at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
                                                                                 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:769)

build.gradle-

android {
    compileSdkVersion 27
    buildToolsVersion '27.0.0'
    defaultConfig {
        applicationId "com.example.mahe.gestureapp"
        minSdkVersion 15
        targetSdkVersion 27
        versionCode 1
        versionName "1.0"
        testInstrumentationRunner "android.support.test.runner.AndroidJUnitRunner"

nisrulz, 2018-10-23:
A couple of things were wrong in your project. I fixed them. here is the zip of the fixed project.

[gestureapp-master.zip](https://github.com/nisrulz/sensey/files/2505684/gestureapp-master.zip)

nisrulz, 2018-10-23:
Closing as it is fixed.

closed 2018-10-23 by nisrulz