nisrulz/sensey #52: ShakeDetion should work in background

pratikbutani, 2019-12-09:
**Do you want to request a *feature* or report a *bug*?**
Feature

**What is the current behavior?**
ShakeDetection is working when app is open. I want it even though app is killed.

**If the current behavior is a bug, please provide the steps to reproduce.**

**Any logs, error output, bugreport etc?**

**What is the expected behavior?**

**Any other comments?**
Please help to work with Shake Sensor in Background too.

**What versions of software are you using?**
+ **Device Information:** 
Samsund Galext J8

+ **Android Version:** 
Android 9.0

nisrulz, 2019-12-09:
This can not be done anymore, from Android 9 onwards. 

Read more: https://developer.android.com/about/versions/pie/android-9.0-changes-all#bg-sensor-access

pratikbutani, 2019-12-12:
There is also written like:

> If your app needs to detect sensor events on devices running Android 9, use a [foreground service.](https://developer.android.com/guide/components/services.html#Foreground)

nisrulz, 2019-12-12:
True, but that is not a implementation I see being bundled in the library. Activity/Services are not supposed to be bundled. They are a client side implementation. The Aim of Sensey is to detect the Gesture, how and where it is needs to be detected is a app developer decision.

pratikbutani, 2019-12-12:
Yeah, you are right. This is really great library. 

Thank you @nisrulz for your time.

closed 2019-12-12 by pratikbutani