nisrulz/sensey #54: 📖 Guest Book 🙇🏻‍♂️

nisrulz, 2020-05-23:
This "issue" serves as a guest book. You're welcome to write nice things about this project, share how it's helped you out, or name drop companies using it.

Idea came from this [tweet](https://twitter.com/wilkieii/status/1262914563521085442).

francogrex, 2022-02-26:
This is a great project, thanks very much for creating it, excellent work. One question: I am wondering for touch events, is it possible to make an even more complex recognition pattern, more so than the long press, swipe and double tap... something like when the user draws a S shape (etc.) for examples it triggers an event?.

nisrulz, 2022-02-27:
> This is a great project, thanks very much for creating it, excellent work. One question: I am wondering for touch events, is it possible to make an even more complex recognition pattern, more so [...]

Please create an issue inside the correct project. This is a different project. I will delete your comment eventually.

nisrulz, 2022-02-27:
Please create a separate issue. This issue is used specifically to collect reviews about the project and not deal with feature requests/bug fixes.

P.S: I will delete your comment here eventually to keep this issue clean and focused on what it was created for.

nisrulz, 2022-02-27:
> This is a great project, thanks very much for creating it, excellent work. One question: I am wondering for touch events, is it possible to make an even more complex recognition pattern, more so [...]

Please create a separate issue. This issue is used specifically to collect reviews about the project and not deal with feature requests/bug fixes.

P.S: I will delete your comment here eventually to keep this issue clean and focused on what it was created for.