nisrulz/sensey #55: feature Update release version

Mir1001, 2020-11-28:
Thanks for the great work. 

I have noticed that the master branch is quite old compared to the development one. Is it possible to make some release upgrade?

nisrulz, 2020-11-29:
Hi @Mir1001 there is no big update that is in the `develop` branch, this is why there has been no release. That branch is basically me doing some cleanup and updating project/sample app but not the library. 

However if it helps, I can make a point release with whatever is in the `develop` branch.

One of things I wanted to do is revamp this library to follow patterns that is common nowadays. For the most parts this library is still useable, but maybe the syntax is not "modern".

Mir1001, 2020-11-29:
Since the last release is 3 years old, I think that the new release will give a signal that this great project is not dead. You can name it minor 1.8.1. For a more "modern" version, you can start with 2.0 ;) Please do it.

closed 2020-11-29 by Mir1001

nisrulz, 2020-12-11:
I have made a new release 1.9.0 🙇🏻‍♂️
I haven't been able to do much testing on my own, so if you find any issues please file them in the issue tracker.