nisrulz/sensey #56: MotionEvent object in callback of gesture/tap detection

jayesh83, 2021-01-06:
Hii thank you so much for such a wonderful library 🔆
I'm using some other library to detect a global touch in my app due to some personal reasons. Which detects touch anywhere in the apo. When there's any touch action I get a callback with motion event. which I can pass to Sensey touchDispatcher. Actually, I've some specific requirements. When there's any callback on gesture detection, say Swipe detection callback. I want MotionEvent object as well, to detect where the event happened. Because now touches are global in the app. I actually don't know on what view or where touch happened. As Sensey currently doesn't provide MotionEvent in callback of detection.

So can we have MotionEvent object in some overridden method? Or how can I get that?

nisrulz, 2021-01-07:
Can you elaborate why you would need the MotionEvent for? Like what data are you trying to extract?

The thing is this is out of the scope of this library. Sensey is a gesture detection library. It will detect a gesture and return a call back with just that. Propagating data through the callback would defeat the goals of this library.

jayesh83, 2021-01-07:
Yaa I get you point. But Nishant problem is that I want to detect MotionEvent.x, MotionEvent.y, MotionEvent.actionType from the click and send it back to the server.

So i'm creating screen sharing and remote access control app where In the clinet side if click event happens I need to send the particular x, y and action back to ther server. and server will then render that touch event on his phone. 

That's why I need to have motionEvent on all the callbacks.

nisrulz, 2021-01-07:
I will take a look if there a possibility of adding a solution, since the public API cannot be changed right away.

Alternatively I think you could just code this functionality using Android api itself. Take a look at how I implemented it in Sensey and create something similar for your usecase. Why I am suggesting this, because your use case is unique. A library usually caters to common solutions and is ideally abstracting away as much complexity it can. Data propagation through callbacks has implications and Sensey wouldn't solve all the problems.

jayesh83, 2021-01-08:
I get your point. Sure then I'll look into Sensey code and will try to create something for my use case. Anyways, thank you so much for being responsive.🔆

jayesh83, 2021-01-08:
Nishant, if possible please create an overridden method where we get MotionEvent object in callback as well. Until then I'll close the issue right away.

nisrulz, 2021-01-08:
I'll keep the issue open, in case someone has the same issue and would like to upvote it. At the same time I would also be looking into your suggestion.

Thank you for your contribution :)

jayesh83, 2021-01-08:
Yeah Sure.